stm32-rs/stm32h7xx-hal #396: ADC circular DMA transfer with multiple channels

ostenning, 2022-10-25:
I'm wondering if the current implementation allows for a circular DMA transfer using multiple channels.

E.g. I basically want to convert this from non-blocking to circular DMA:

```
        let a: u32 = self.adc.read(&mut self.pins.0).unwrap();
        let b: u32 = self.adc.read(&mut self.pins.1).unwrap();
        let c: u32 = self.adc.read(&mut self.pins.2).unwrap();
```

As I understand only one buffer can be nominated for the transfer, I would be happy if the solution could utilize some kind of round-robin sharing of one buffer - if that's feasible. 

Any assistance appreciated, once again!

Thanks

richardeoin, 2022-10-25:
This sounds very useful, but I don't think it's supported currently. It would need a new version of the [start_conversion_dma](https://docs.rs/stm32h7xx-hal/latest/stm32h7xx_hal/adc/struct.Adc.html#method.start_conversion_dma) method to setup the ADC peripheral to setup round-robin sampling

dobrite, 2023-07-28:
I'm unsure if [this draft PR](https://github.com/stm32-rs/stm32h7xx-hal/pull/442) is exactly what this improvement had in mind, but it definitely allows for circular DMA transfer using multiple channels!

I'm looking for feedback for improvements, so if y'all have any, feel free to comment!