stm32-rs/stm32h7xx-hal #401: I2C WriteRead() and Read() do not wait for I2C stop condition.

nkrackow, 2022-11-03:
An I2C transaction is completed by the [stop condition](https://www.analog.com/en/technical-articles/i2c-primer-what-is-i2c-part-1.html#:~:text=not%20acknowledge%20bits.-,Start%20and%20Stop%20Conditions,-As%20the%20name). 
This condition does not happen when many `WriteRead()` or `Read()` transactions happen right after the another. The I2C bus goes strait into the next transaction with the next start condition. This does not comply with the [embedded hal](https://github.com/rust-embedded/embedded-hal/blob/master/embedded-hal/src/i2c.rs#L319-L340) and can lead to undefined behavior (observed with a MCP23017 chip).

jordens, 2022-11-03:
Hypothesis is that the CPU is fast enough after [reading the last last byte](https://github.com/stm32-rs/stm32h7xx-hal/blob/21d9c06a57169e92cf5745e582dca171c73294c9/src/i2c.rs#L631) to [set new nbytes and start the next transfer](https://github.com/stm32-rs/stm32h7xx-hal/blob/21d9c06a57169e92cf5745e582dca171c73294c9/src/i2c.rs#L570) while the peripheral takes its time generating the last NAK so that it skips generating the STOP and instead proceeds to generate another (RE)START. Maybe also grey zone between silicon bug and RM inaccuracies.
In any case the transaction is not completed as demanded by the HAL.
Would be really great if we could cut a 0.13.1 for this.

richardeoin, 2022-11-03:
Thanks very much for reporting! I'm definitely open to cutting a 0.13.1 for this.

This is supposed to work by the peripheral issuing the stop condition automatically, but if that's not occurring in some situations it needs a fix. I'll review #402 in the coming days

ryan-summers, 2022-11-04:
The CPU can likely very easily race (and most likely win) against the I2C bus. The I2C bus is running at a nominal 100KHz, whereas the CPU is often running in the 400MHz range. There is ample time to do (many) other tasks before a single I2C clock cycle occurs. We definitely need to be waiting for the stop to transmit in these functions before returning.

closed 2022-11-04 by ryan-summers

richardeoin, 2022-11-06:
I'll push the fix for this as 0.13.1 shortly 👍

jordens, 2022-11-07:
@richardeoin Thanks!

PR #402 (2022-11-04): Bugfix for I2C transaction stop conditions
Simply wait until the busy flag is released. Now the stop condition occurs (right before the next start condition in the case of successive writereads). 
Fixes #401