stm32-rs/stm32h7xx-hal #403: Not possible to set SYSCLK to 480 MHz on STM32H743 (panics with 'assertion failed: vco_ck <= vco_max')

Jonarw, 2022-11-14:
Let me start by stating that I am quite new to `stm32h7xx-hal` and Rust itself, so if I have overlooked something obvious, please excuse me.

I am trying to run my STM32H743 at its maximum frequency of 480 MHz. This works in my C project with the code generated by STM32CubeMX. This is the relevant section of the clock setup in CubeMX:
[image]

I would essentially like to recreate that in my Rust project. I am doing the following:

        let rcc = cx.device.RCC.constrain();

        let pwr = cx.device.PWR.constrain();
        let pwrcfg = pwr.freeze();
    
        let ccdr = rcc
            .sys_ck(480.MHz())
            .pll1_q_ck(40.MHz())
            .freeze(pwrcfg, &cx.device.SYSCFG);

However, when running this I get the following error message:
`ERROR panicked at 'assertion failed: vco_ck <= vco_max', C:\...\github.com-1ecc6299db9ec823\stm32h7xx-hal-0.13.1\src\rcc\pll.rs:363:5`

I can't see a reason why this shouldn't work. It does work just fine if I specify `.sys_ck(400.MHz())` instead.
I assume the error must occur in here:
https://github.com/stm32-rs/stm32h7xx-hal/blob/7b465225b0ed2350b6b0933b98d673645790272a/src/rcc/pll.rs#L48
However I am not very comfortable around Rust macros yet, so I am not quite sure how to step through with the debugger and understand why the assertion fails...

Am I doing something wrong? Or is this a bug?

richardeoin, 2022-11-14:
Hi @Jonarw ! Thanks for reporting the issue. This probably isn't a bug, but rather the crate's documentation needs to improved to highlight this often-used feature of the H7.

There's two things you need to do to make this work:
* Enable Voltage Scaling 0 (VOS0) for frequencies above 400MHz. To achieve 480MHz the device voltage needs to be increased, and there's a [special option](https://docs.rs/stm32h7xx-hal/latest/stm32h7xx_hal/pwr/struct.Pwr.html#method.vos0) for that.
* Use the VCOH oscillator inside the PLL. The default is VCOL, which has a maximum frequency of 420MHz. You can switch by using the Iterative PLL strategy. That's specified with a builder method like `rcc.pll1_strategy(rcc::PllConfigStrategy::Iterative)`

There's an example [here](https://github.com/stm32-rs/stm32h7xx-hal/blob/7b465225b0ed2350b6b0933b98d673645790272a/examples/vos0.rs) that does both of those things. Does that help you get this working?

mattico, 2022-11-15:
Also make sure you enable the cargo feature `stm32h743v`. Emphasis on the **v**.

Jonarw, 2022-11-15:
Yes, that immediately solved the problem, thanks for pointing me in the right direction!

And I would like to add that I am very impressed so far with the embedded infrastructure in Rust in general, and with this project in particular! The HAL actually provides a useful abstraction of the hardware, which is something that ST did not quite manage with their C HAL in my opinion. Much appreciated!

closed 2022-11-15 by Jonarw