stm32-rs/stm32h7xx-hal #405: Why no HAL_PCDEx_PMAConfig for H7xx?

davidedelvento, 2022-11-26:
I'm trying to port some (MIDI) USB code written to make a STM32f1xx as USB device. I need the code to run on a STM32h743. The first board invokes `HAL_PCDEx_PMAConfig` during initialization, but of course there isn't such a thing for h7xx.

Why is it so, and what to do instead?

As a first try, I just avoided the call and the H743 USB device is recognized as configured, but no data can be transferred. Invocations to `USBD_LL_Transmit` on the board return `USBD_OK` but no data is ever received by the host.

If relevant, this is with a Nucleo-H743ZI2

mattico, 2022-11-28:
It sounds like you're using ST's HAL. Can you explain how you're using that with this project?

davidedelvento, 2022-11-29:
Thanks for getting back to me.

I am trying to port this project: https://github.com/Hypnotriod/midi-box-stm32 @Hypnotriod 
After some trying, and without calling `HAL_PCDEx_PMAConfig`, I eventually got it to work. So to answer my own question, that function is not needed for h7xx and no need to do anything instead (not sure if this is true in general, but it seems to be the case for what I am doing).

I guess my main gripe is how much different are the APIs for doing the same thing (in this case simply initializing USB) between "very similar" boards (or maybe the gripe is that these boards should not be treated as "very similar", e.g. by marketing and sales)

richardeoin, 2022-12-05:
Thanks for the discussion. This question seems to be resolved so I will close the issue.

closed 2022-12-05 by richardeoin