stm32-rs/stm32h7xx-hal #417: usb_passthrough example does not compile due to broken USB2 implementation

cosmikwolf, 2023-02-04:
```
error: Cannot not select both rm0433 and rm0399
   --> src/lib.rs:131:1
    |
131 | compile_error!("Cannot not select both rm0433 and rm0399");
    | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```

cosmikwolf, 2023-02-04:
Here is an example where I try and implement USB2. This example compiles and runs, however I am not able to use the USB2 port on an STM32H750v dev board
[https://gist.github.com/cosmikwolf/f2ff77bc6e4d32e7ef8b07f859c6185d](url)

mattico, 2023-02-10:
When you enable the chip feature `stm32h750v` [it enables the appropriate `rmXXXX` feature automatically](https://github.com/stm32-rs/stm32h7xx-hal/blob/87448d28f348aef6622ff1667885236ec95ac163/Cargo.toml#LL117-L117C10). So, remove the `rm0399` feature. Is there a reason that you added that feature?

closed 2024-12-30 by richardeoin