stm32-rs/stm32h7xx-hal #424: Creation of SAI pin tuple without I2SPinSdB

JoseGuilhermeCR, 2023-02-27:
Hi there,

I'm trying to use the SAI peripheral on an H735... But I'm stuck as I don't have an SdB pin and using `None` on the tuple will lead to `rustc E0282` and `rustc E0283`.

```
let sai1_pins = (
            mclk_a_pin.into_alternate(),
            sck_a_pin.into_alternate(),
            fs_a_pin.into_alternate(),
            sd_a_pin.into_alternate(),
            None
        );
```

The compiler says I should say what Option<T> none belongs to... So I thought something like this would work, but now I've got another compiler error, `rustc E0277`.
```
let sai1_pins = (
            mclk_a_pin.into_alternate(),
            sck_a_pin.into_alternate(),
            fs_a_pin.into_alternate(),
            sd_a_pin.into_alternate(),
            None::<sai::i2s::I2SPinSdB<pac::SAI1>>
        );
```

The SAI example available in the `examples` directory uses SdB pin... but it's not shown anywhere how `None` should be used in the case you don't have this pin. 

The log of the first error is as follows:

```
error[E0283]: type annotations needed
   --> src/pcm_4202_driver.rs:62:13
    |
62  |             None
    |             ^^^^ cannot infer type of the type parameter `T` declared on the enum `Option`
...
65  |         let mut sai1 = sai1.i2s_ch_a(
    |                             -------- type must be known at this point
    |
    = note: multiple `impl`s satisfying `_: sai::i2s::I2SPinSdB<stm32h7xx_hal::stm32::SAI1>` found in the `stm32h7xx_hal` crate:
            - impl sai::i2s::I2SPinSdB<stm32h7xx_hal::stm32::SAI1> for stm32h7xx_hal::gpio::Pin<'E', 3, stm32h7xx_hal::gpio::Alternate<6>>;
            - impl sai::i2s::I2SPinSdB<stm32h7xx_hal::stm32::SAI1> for stm32h7xx_hal::gpio::Pin<'F', 6, stm32h7xx_hal::gpio::Alternate<6>>;
    = note: required for `(stm32h7xx_hal::gpio::Pin<'E', 2, stm32h7xx_hal::gpio::Alternate<6>>, ..., ..., ..., ...)` to implement `sai::i2s::I2SPinsChA<stm32h7xx_hal::stm32::SAI1>`
    = note: the full type name has been written to 'target/thumbv7em-none-eabihf/debug/deps/tcc-9b1e8f6e0ab9f3e3.long-type-11402833594598164313.txt'
note: required by a bound in `i2s_ch_a`
   --> .cargo/registry/src/github.com-1ecc6299db9ec823/stm32h7xx-hal-0.13.1/src/sai/i2s.rs:359:15
    |
359 |         PINS: I2SPinsChA<Self>;
    |               ^^^^^^^^^^^^^^^^ required by this bound in `i2s_ch_a`
help: consider specifying the generic argument
    |
62  |             None::<T>
    |                 +++++
```

Now I'm definitely not an expert in the language, I've only used it for a few years now, but not professionally nor with this heavy usage of templates and etc... Am I missing something really stupid?

JoseGuilhermeCR, 2023-03-05:
I got rid of the error by using `None::<gpio::Pin<'E', 3, Alternate<6>>>`, but it's *definitely* ugly... Still think there might be a way where I don't need to concern myself over the port/pin/mode of something I won't use.

burrbull, 2023-04-17:
https://github.com/stm32-rs/stm32h7xx-hal/pull/431 does not change Sai interface, but also support pins in Default (`Analog`) mode to be passed in constructor.
So this also should work:
```rust
let sai1_pins = (
    mclk_a_pin,
    sck_a_pin,
    fs_a_pin,
    sd_a_pin,
    None::<gpio::PE3>
);
```