stm32-rs/stm32h7xx-hal #425: sdmmc_fat does not compile

Orange-Murker, 2023-03-11:
The `sdmmc_fat` example does not compile with the following error:

```
error[E0277]: the trait bound `SdmmcBlockDevice<Sdmmc<SDMMC2, SdCard>>: Deref` is not satisfied
  --> src/main.rs:92:24
   |
92 |     let mut sd_fatfs = Controller::new(sd.sdmmc_block_device(), TimeSource);
   |                        ^^^^^^^^^^^^^^^ the trait `Deref` is not implemented for `SdmmcBlockDevice<Sdmmc<SDMMC2, SdCard>>`
   |
   = help: the trait `BlockDevice` is implemented for `BlockSpi<'_, SPI, CS>`
   = note: required for `SdmmcBlockDevice<Sdmmc<SDMMC2, SdCard>>` to implement `BlockDevice
```

olback, 2023-03-12:
I can't seem to reproduce the issue. What processor are you targeting?
```
git clone https://github.com/stm32-rs/stm32h7xx-hal
git checkout v0.13.1
cargo build --example sdmmc_fat --features stm32h747cm7,sdmmc,sdmmc-fatfs
```

Orange-Murker, 2023-03-12:
I am targeting STM32H7B0

Orange-Murker, 2023-03-12:
> I can't seem to reproduce the issue. What processor are you targeting?
> 
> ```
> git clone https://github.com/stm32-rs/stm32h7xx-hal
> git checkout v0.13.1
> cargo build --example sdmmc_fat --features stm32h747cm7,sdmmc,sdmmc-fatfs
> ```

I haven't managed to reproduce the issue with those steps as well.

To reproduce the issue I created a new project with the following dependencies in my `Cargo.toml`:
```toml
[dependencies]
cortex-m = "0.7.4"
cortex-m-rt = "0.7.1"
stm32h7xx-hal = {version = "0.13.1", features = ["stm32h743v", "rt", "sdmmc", "sdmmc-fatfs"]}
embedded-sdmmc = "0.4.0"
panic-halt = "0.2.0"
```
I have also set up the memory.x and config.toml as per usual.

I then copied the sdmmc_fat.rs file into main and got rid of all the stuff related to utilities and added a panic handler.
The program did not compile.

Orange-Murker, 2023-03-12:
Here is the full code. Everything is identical to the original example except for the fact that the utilities are not used.

```rust
#![no_main]
#![no_std]

use {
    embedded_sdmmc::{Controller, VolumeIdx},
    stm32h7xx_hal::sdmmc::{SdCard, Sdmmc},
    stm32h7xx_hal::{pac, prelude::*, rcc},
};

extern crate panic_halt;

// This is just a placeholder TimeSource. In a real world application
// one would probably use the RTC to provide time.
pub struct TimeSource;

impl embedded_sdmmc::TimeSource for TimeSource {
    fn get_timestamp(&self) -> embedded_sdmmc::Timestamp {
        embedded_sdmmc::Timestamp {
            year_since_1970: 0,
            zero_indexed_month: 0,
            zero_indexed_day: 0,
            hours: 0,
            minutes: 0,
            seconds: 0,
        }
    }
}

#[cortex_m_rt::entry]
unsafe fn main() -> ! {

    // Get peripherals
    let cp = cortex_m::Peripherals::take().unwrap();
    let dp = pac::Peripherals::take().unwrap();

    // Constrain and Freeze power
    let pwr = dp.PWR.constrain();
    let pwrcfg = pwr.freeze();

    // Constrain and Freeze clock
    let ccdr = dp
        .RCC
        .constrain()
        .sys_ck(200.MHz())
        .pll1_strategy(rcc::PllConfigStrategy::Iterative)
        .pll1_q_ck(100.MHz())
        .pll2_strategy(rcc::PllConfigStrategy::Iterative)
        .pll3_strategy(rcc::PllConfigStrategy::Iterative)
        .freeze(pwrcfg, &dp.SYSCFG);

    // Get the delay provider.
    let mut delay = cp.SYST.delay(ccdr.clocks);

    let gpiob = dp.GPIOB.split(ccdr.peripheral.GPIOB);
    let gpiod = dp.GPIOD.split(ccdr.peripheral.GPIOD);

    let mut sd: Sdmmc<_, SdCard> = dp.SDMMC2.sdmmc(
        (
            gpiod.pd6.into_alternate(),
            gpiod.pd7.into_alternate(),
            gpiob.pb14.into_alternate(),
            gpiob.pb15.into_alternate(),
            gpiob.pb3.into_alternate(),
            gpiob.pb4.into_alternate(),
        ),
        ccdr.peripheral.SDMMC2,
        &ccdr.clocks,
    );

    // Loop until we have a card
    loop {
        // On most development boards this can be increased up to 50MHz. We choose a
        // lower frequency here so that it should work even with flying leads
        // connected to a SD card breakout.
        match sd.init(2.MHz()) {
            Ok(_) => break,
            Err(err) => {
            }
        }

        delay.delay_ms(1000u32);
    }

    // See https://github.com/rust-embedded-community/embedded-sdmmc-rs for docs
    // and more examples

    let mut sd_fatfs = Controller::new(sd.sdmmc_block_device(), TimeSource);
    let sd_fatfs_volume = sd_fatfs.get_volume(VolumeIdx(0)).unwrap();
    let sd_fatfs_root_dir = sd_fatfs.open_root_dir(&sd_fatfs_volume).unwrap();
    sd_fatfs
        .iterate_dir(&sd_fatfs_volume, &sd_fatfs_root_dir, |entry| {
        })
        .unwrap();
    sd_fatfs.close_dir(&sd_fatfs_volume, sd_fatfs_root_dir);

    loop {
        cortex_m::asm::nop()
    }
}
```

olback, 2023-03-13:
Ah, I've figured it out...
Version `0.13.1` of this crate uses `embedded-sdmmc` version `0.3.0`. `embedded-sdmmc` has been updated in the master branch but not yet released, `embedded-sdmmc` version `0.4.0` will land in version `0.14.0` (Is this true? @richardeoin).

The error message rustc produces is complete garbage though.

Orange-Murker, 2023-03-13:
Thank you for figuring it out!
I was really puzzled by that error message.

closed 2023-03-13 by Orange-Murker

richardeoin, 2023-03-13:
Great, nice investigation @olback! This "multiple versions of an upstream crate" issue has also happened to me in the past. The [cargo tree](https://doc.rust-lang.org/cargo/commands/cargo-tree.html) subcommand is quite useful for investigating, but I'm not aware of a solution with better diagnostics :(

Indeed the master branch of this crate has been updated to depend on `embedded-sdmmc` version `0.4.0` already (by @olback !) and that will be included in the next release to crates.io