stm32-rs/stm32h7xx-hal #433: WindowWatchdogReset isn't matched by ResetReason properly despite valid RCC_RSR

phansel, 2023-06-14:
A system window watchdog reset results instead in Unknown { rcc_rsr: 268435456 }, which is just 2^28 i.e bit 28 (WWDG1RSTF) high, all others low. It should be returning WindowWatchdogReset per https://github.com/stm32-rs/stm32h7xx-hal/blob/master/src/rcc/reset_reason.rs#L44.  

There's also a spelling error in the above lines of src/rcc/reset_reason.rs: `is_reset_occourred()` should be `is_reset_occurred()`. I'm not able to locate where that function originates, though.

To reproduce, modify the reset_reason.rs example:
line 31
+ let mut watchdog = system_watchdog::SystemWindowWatchdog::new(dp.WWDG, &ccdr);
line 48
+ watchdog.start(100.millis());

Device: STM32H743ZI2 (stm32h743v)
stm32h7xx-hal version: 0.14.0

richardeoin, 2023-06-15:
Hi @phansel ! Thanks for reporting.

The `is_reset_occourred()` function originates from the [Peripheral Access Crate (PAC)](https://github.com/stm32-rs/stm32-rs). There's more information in that repository, but basically that function is actually autogenerated based on [this yaml file](https://github.com/stm32-rs/stm32-rs/blob/master/peripherals/rcc/rcc_v3.yaml#L124). The spelling would need to be fixed there first, and then updated here. 

The bug here seems to be that this code was expecting the only valid RSR combinations are those in this table in RM0433. You've found one that has only a 1 in the WWDG column (and zeros elsewhere), which wasn't expected.
[image]

I have no idea why the '1's in row 6 of the table are bold however. That might mean something? I share your opinion that the combination you've found (only WWDG1RSTF set) seems valid though. This code could cover it by adding another arm to the match statement.

phansel, 2023-06-17:
@richardeoin Thanks for the reviews. I've tackled the primary issue in #434. The typo can be fixed later; thanks for the pointer.

phansel, 2023-06-22:
@richardeoin Did you have the chance to review #434 ?

richardeoin, 2023-07-01:
Yes - reviewed now. I will close the issue now that #434 is merged

closed 2023-07-01 by richardeoin