stm32-rs/stm32h7xx-hal #435: i2c can lock up MCU

ostenning, 2023-06-20:
I've experienced a write to the i2c peripheral locking up the MCU and not returning any result, this can happen for example if the device it is connected to isn't powered up. I would expect that writing to the i2c peripheral should return an Error in this scenario.

I've also experienced this when attempting to implement BDMA and implement multiple successive i2c.write and then i2c.master_re_start commands, as adapted from the [i2c bdma example](https://github.com/stm32-rs/stm32h7xx-hal/blob/master/examples/i2c4_bdma.rs). 

After successfully receiving data via `master_re_start` and clearing appropriate interrupt flags, then on the second i2c.write, the function never returns and the MCU locks up.

cc @richardeoin

richardeoin, 2023-07-02:
I'm aware this can happen, but I don't believe it is a defect in the HAL implementation. 

> the device it is connected to isn't powered up

It sounds like both lines on your I2C bus are low. The [I2C specification](https://www.nxp.com/docs/en/user-guide/UM10204.pdf) states "When the bus is free, both lines are HIGH". Therefore the defect is in your bus because it is not complying to the specification.

For some applications handling non-confirming or defective busses is useful or even essential, but imo this functionality belongs in a separate crate.

richardeoin, 2023-07-02:
> implement multiple successive i2c.write and then i2c.master_re_start commands, as adapted from the [i2c bdma example](https://github.com/stm32-rs/stm32h7xx-hal/blob/master/examples/i2c4_bdma.rs).

This sounds like a different problem. Can you provide code that triggers this? Then I can keep the issue open

closed 2023-11-20 by ostenning