stm32-rs/stm32h7xx-hal #446: USB Host example

zUnixorn, 2023-08-08:
Is it currently possible to use the USB-OTG in host mode with this hal crate? The examples only show it in device mode and I cannot find anything in the docs imlying that.

richardeoin, 2023-08-10:
Hey! No it's not possible to use the USB in host mode with this HAL crate :(

It would need to be implemented in [synopsys-usb-otg](https://github.com/stm32-rs/synopsys-usb-otg) or a similar crate, which is not the case currently. Someone already made [an issue](https://github.com/stm32-rs/synopsys-usb-otg/issues/20) there about it

Dirbaio, 2023-09-12:
There's one implementation here https://github.com/nviennot/turbo-resin/tree/main/src/drivers/usb (Apache+MIT)