stm32-rs/stm32h7xx-hal #448: regression in ethernet DMA initialization for H743 chips

ritikmishra, 2023-08-25:
One of the changes merged in by #445 moves the write to the ETH_DMAMR register to before the ethernet peripheral clocks are enabled

https://github.com/stm32-rs/stm32h7xx-hal/pull/445/files#diff-bc6d950ee1ed8e0df070d5ac2e9190dea3730fc251d9ca1b3c9bdad8c7f0bd3aR458-L478

This breaks the ethernet DMA initialization on my STM32H743IIT6

ryan-summers, 2023-08-28:
I can confirm that Stabilizer's hardware-in-the-loop tests indicate that networking no longer functions on the H743 chips [with the latest hal rev](https://github.com/quartiq/stabilizer/pull/774#event-10206101055)

richardeoin, 2023-09-11:
I can reproduce this on two development boards:
* STM32H747I-DISCO (STM32H747XIH)
* STM32H735G-DK (STM32H735IGK)

For me the symptom is that the PHY never reports the link is up (over the MDIO interface) and the board does not respond to pings. Reverting the change to `eth.rs` from #445 resolves the issue.

Why the change in #445 is required for the NUCLEO-H723ZG but fails on the STM32H735G-DK is really not clear. Those STM32 part share a reference manual, so I would expect them to both work with the same initialisation sequence. Perhaps either the board power configuration (LDO vs SMPS) or PHY configuration is to blame.

In the meantime I would propose reverting the change to `eth.rs` from #445 until we understand why it is needed for the NUCLEO-H723ZG.

ryan-summers, 2023-09-11:
I'm on board with your assessment @richardeoin - I didn't see a thorough explanation in #445 why it was required outside of "Seems to work now". If that truly is the case, we can device gate it, but feels like more investigation is required.

PR #450 (2023-09-12): Revert "Move ethernet DMA reset slightly earlier to eliminate freeze on h723"
Revert the changes to `eth.rs` from #445 

Fixes #448
closed 2023-09-12 by richardeoin