stm32-rs/stm32h7xx-hal #463: Reading internal flash triggers HardFault

HaoboGu, 2023-10-11:
Hello, I'm using stm32h7b0 mcu and trying to emulate eeprom using mcu's flash and `stm32h7xx-hal::flash`. Here is my testing code for flash write/read: 

```rust
        let (mut flash, _) = dp.FLASH.split();
        let mut bytes = [8u8;16];
        const FLASH_SECTOR_15_ADDR: u32 = 15 * 8192;
        let sector = flash.sector(FLASH_SECTOR_15_ADDR as usize).unwrap();
        info!("{}", sector.offset);
        if sector.contains(FLASH_SECTOR_15_ADDR as usize) {
            embedded_storage::nor_flash::ReadNorFlash::read(&mut flash, FLASH_SECTOR_15_ADDR, &mut bytes).unwrap();
        }
```

But the `read` function fails and the program jumps to HardFault 
[image]

Here is the callstack: 

[image]

richardeoin, 2023-10-11:
Hi! Thanks for reporting. The stm32h7b0 is a device with only 128KB of flash, divided into 16 sectors of 8KB each. You're trying to read from the last of those sectors. In theory that should be possible.

Could you confirm:
* That you're using the `--features=stm32h7b0` feature flag to build your test code.
* If possible, the value returned by `stm32h7xx_hal::signature::FlashSize::bytes()` on your stm32h7b0.

HaoboGu, 2023-10-12:
Yes I'm reading the last sector of the flash. 

I checked my config, here is how i introduce `stm32h7xx-hal` in `Cargo.toml`:

```toml
stm32h7xx-hal = { version = "0.14.0", features = [
    "stm32h7b0",
    "rt",
    "log-rtt",
    "usb_hs",
] }
```

The value returned by `stm32h7xx_hal::signature::FlashSize::bytes()` is 131072, aka 128kb:
[image]

I've noticed there is a new version of `stm32h7xx-hal` published at crates.io, I also tried it, but my code breaks at:

```
note: required by a bound in `gpio::convert::<impl stm32h7xx_hal::gpio::Pin<P, N, MODE>>::into_alternate`
   --> /Users/haobogu/.cargo/registry/src/rsproxy.cn-0dccff568467c15b/stm32h7xx-hal-0.15.0/src/gpio/convert.rs:151:15
    |
147 |     pub fn into_alternate<const A: u8>(
    |            -------------- required by a bound in this associated function
...
151 |         Self: marker::IntoAf<A>,
    |               ^^^^^^^^^^^^^^^^^ required by this bound in `gpio::convert::<impl Pin<P, N, MODE>>::into_alternate`

error[E0277]: the trait bound `stm32h7xx_hal::gpio::Pin<'A', 12>: gpio::marker::IntoAf<10>` is not satisfied
   --> boards/stm32h7/src/main.rs:83:33
    |
83  |         let usb_dp = gpioa.pa12.into_alternate();
    |                                 ^^^^^^^^^^^^^^ the trait `gpio::marker::IntoAf<10>` is not implemented for `stm32h7xx_hal::gpio::Pin<'A', 12>`
```

HaoboGu, 2023-10-12:
I tried to erase the sector using: 

```rust
let (mut flash, _) = dp.FLASH.split();
let mut unlocked = flash.unlocked();
unlocked.erase_sector(15).unwrap();
```

The program reported the following error:

```
panicked at boards/stm32h7/src/main.rs:101:35:
called `Result::unwrap()` on an `Err` value: EccDoubleDetection
```

It seems that some of my write operations didn't complete, and then an ECC error occured.

@richardeoin  Does `stm32h7xx-hal` crate provide any methods to fix flash ECC error?

mlamoore, 2023-10-12:
Assuming the flash is fine and just got interrupted while writing or
something, shouldn't a flash erase and write fix the ECC error?

HaoboGu, 2023-10-12:
I've reviewed my code. I use the following function to write the flash:

```rust
fn write_record(&mut self, record: EepromRecord) {
        match self
            .storage
            .write(STORAGE_START_ADDR + self.pos, &record.to_bytes())
        {
            Ok(_) => self.pos += 4,
            Err(_) => error!("Failed to write record to storage"),
        }
    }
```

after each write, the next write addr will be addr + 4. Is this right way to use the crate? 

I believe I didn't write much data(about 300 bytes). So I don't know is the flash still fine. Erasing the sector with `unlocked.erase_sector(15).unwrap();`, reports `EccDoubleDetection` error. Does it mean the flash has been broken?

richardeoin, 2023-10-22:
Your report about USB is a separate issue. I've posted a fix in https://github.com/stm32-rs/stm32h7xx-hal/pull/464

HaoboGu, 2023-10-27:
Erasing the sector fixed the ECC error, closing this issue

closed 2023-10-27 by HaoboGu