stm32-rs/stm32h7xx-hal #466: nukleo-h723zg  usb  building problem

alevstiv, 2023-10-30:
need usb full speed on h723zg 
1) cargo build --features=stm32h735,rt,usb_hs, gpio-h747 --example usb_serial  -  building  with compilation error( and no  build option for gpio-h72 )
2) as a separate project ( stm32h7xx-hal = {version = "0.15.0", features = ["stm32h735","rt","usb_hs"]}) generate compilation errors: 
error[E0425]: cannot find value `pin_dm` in this scope
  --> src/main.rs:67:9
   |
67 |         pin_dm,
   |         ^^^^^^ not found in this scope

error[E0425]: cannot find value `pin_dp` in this scope
  --> src/main.rs:68:9
   |
68 |         pin_dp,
   |         ^^^^^^ not found in this scope

3)  if you comment out  #[cfg(any(feature = "rm0455", feature = "rm0468"))] then:

error[E0277]: the trait bound `stm32h7xx_hal::gpio::Pin<'A', 12>: gpio::marker::IntoAf<10>` is not satisfied
   --> src/main.rs:61:50
    |
61  |         (gpioa.pa11.into_alternate(), gpioa.pa12.into_alternate())
    |                                                  ^^^^^^^^^^^^^^ the trait `gpio::marker::IntoAf<10>` is not implemented for `stm32h7xx_hal::gpio::Pin<'A', 12>`

richardeoin, 2023-10-30:
Hi, thanks for reporting. This should already be fixed in https://github.com/stm32-rs/stm32h7xx-hal/pull/464 . I'll push this as a patch release 0.15.1 shortly.

alevstiv, 2023-10-31:
Thanks!  It works.
But with settings 
    stm32h7xx-hal = {version = "0.15.1", path = "../stm32h7xx-hal-0.15.1", features = ["stm32h735","rt","usb_hs"]}
and enable  #[cfg(any(feature = "rm0455", feature = "rm0468"))] results in a compilation error:
    "pin_dm, | ^^^^^^ not found in this scope"
how to configure it correctly?

closed 2023-11-02 by alevstiv