stm32-rs/stm32h7xx-hal #472: How to perform SPI write before starting DMA transfer?

nullobject, 2023-11-29:
Is it possible to do a normal SPI write before starting a DMA transfer?

For example, in the [SPI RTIC example](https://github.com/stm32-rs/stm32h7xx-hal/blob/master/examples/spi-dma-rtic.rs#L160-L170) it would be nice to be able to do something like this:

```rust
transfer.start(|spi| {
    // Write a command before beginning the transfer
    cs.set_low();
    dc.set_low();
    spi.write(&[0x11u8, 0x22, 0x33]);
    cs.set_high();
    dc.set_high();

    // Write frame buffer using DMA transfer
    cs.set_low();
    spi.enable_dma_tx();
    spi.inner_mut().cr1.modify(|_, w| w.spe().enabled());
    spi.inner_mut().cr1.modify(|_, w| w.cstart().started());
});
```

I am trying to use DMA to write the frame buffer to an OLED display (SSD1322), but I need to write a couple of commands to the display before sending the data.

Any suggestions?

mlamoore, 2023-11-29:
I'm this situation, why wouldn't you just use DMA to send the command as
well as the data? It seems simpler to me (in C as well as Rust) to set up
the SPI and DMA peripherals to work together, then send one short DMA
transfer with the command, followed by a longer DMA transfer with the data.

nullobject, 2023-11-29:
> I'm this situation, why wouldn't you just use DMA to send the command as
> 
> well as the data? It seems simpler to me (in C as well as Rust) to set up
> 
> the SPI and DMA peripherals to work together, then send one short DMA
> 
> transfer with the command, followed by a longer DMA transfer with the data.

@mlamoore Thanks for the reply.

I need to send 3 commands to the display (row address, column address, write enable) before sending the frame buffer data.

Sending the commands and frame buffer with DMA would require extra state so the interrupt handler knows what to do next. e.g. The DC signal needs to be asserted for data, and deasserted for commands. The CS signal needs to be asserted/deasserted for each transaction.

In C, this is three blocking `HAL_SPI_Transmit` calls followed by a `HAL_SPI_Transmit_DMA` call, which is quite simple to implement.

nullobject, 2024-01-09:
@mlamoore Hi again. I am still looking for a solution.

Do you know of any examples of how to do multiple DMA transfers, like you suggested?

Does this require freeing each DMA transfer when it is complete to gain access to the resources (i.e. SPI instance) and start another transfer?