stm32-rs/stm32h7xx-hal #485: 480MHz and vos0 unstable on stm32h747XI (Arduino GIGA R1 WiFi)

bsodmike, 2024-02-06:
Hi all,

I've been trying to get 480MHz & vos0 working on the above board (see https://github.com/bsodmike/h7/blob/master/h7-cm7/src/main.rs#L52-L115) however, this is unstable and when I power-cycle my board it just bricks itself.

However, a lower clock speed works just fine,

```rust
    let mut pwrcfg = pwr.ldo().vos1().freeze();

    let rcc = dp.RCC.constrain();
    let ccdr = rcc.sys_ck(100.MHz()).freeze(pwrcfg, &dp.SYSCFG);
```

Would appreciate if anyone has any pointers in getting this to run at 480MHZ, thanks!

olback, 2024-02-06:
I'm not familiar with that particular board but looking at the schematic, it appears to have an external 16MHz crystal. 
[image]

Something like this should work I think:
```rust
let mut ccdr = dp
    .RCC
    .constrain()
    .use_hse(HSE_CLK)
    // .bypass_hse() // probably don't want this? 
    .sys_ck(480.MHz())
    .hclk(240.MHz())
    // ... 
    .freeze(pwrcfg, &dp.SYSCFG);
```

bsodmike, 2024-02-06:
Hi @olback nice to meet you! I really liked your h7 project which helped me a lot to get started.  Ooh this makes total sense, will try it!!

Thank you, M

bsodmike, 2024-02-07:
@olback as per your suggestion, I tried replacing https://github.com/bsodmike/h7/blob/master/h7-cm7/src/main.rs#L102 with `.use_hse(16.MHz())` - no other changes.  I had to re-flashed the Mbed OS to get a working "power stage" loaded so that the board would work at all.

Flashing the amended h7-cm7 firmware would cause the blue led to flash (https://github.com/bsodmike/h7/blob/master/h7-cm7/src/main.rs#L511-L515).  However, if I power-cycle the board it "freezes" up again.

I also find my Jlink-mini has trouble connecting to the board at times; it seems to be rather sporadic.

bsodmike, 2024-02-25:
I managed to get this to [work with embassy-stm32](https://github.com/bsodmike/arduino-giga-r1-wifi-stm32h747xi-async-quickstart/blob/master/src/board.rs#L98-L124) with similar config.  If anyone figures this out, feel free to comment but I'm closing this.

closed 2024-02-25 by bsodmike