stm32-rs/stm32h7xx-hal #499: STM32H725 can't run at 550MHz

ilmai, 2024-06-04:
I'm trying to set the system clock of STM32H752 to 550Mhz but getting an assertion failure:

`assertion failed: sys_d1cpre_ck <= sys_d1cpre_ck_max`

It seems like the maximum frequency of the 725 is set at 520MHz for VOS0 on line 748 of `rcc/mod.rs`, which doesn't match for example what Cube IDE specifies as the maximum. I just started using the HAL today so I might be missing something, but this looks wrong to me?

richardeoin, 2024-06-04:
Hey, thanks for reporting! The maximum CPU frequency is 520MHz unless bit `CPU_FREQ_BOOST` is set in register SYSCFG_UR18. The HAL implementation assumes that bit is not set. It's possible that CubeMX generates code to set that bit, which allows it to offer the higher speed (with the downside that you loose ECC!)

I think the HAL should document this better. It is mentioned [here,](https://docs.rs/stm32h7xx-hal/latest/stm32h7xx_hal/pwr/index.html#boost-mode-vos0) but you might not have found that section easily. Ideas about improving the documentation are very welcome!

ilmai, 2024-06-05:
Thanks for the quick response! I have to say I really appreciate how straightforward the clock API is, even if some details like this are a bit obscure coming from Cube.

closed 2024-06-05 by ilmai