stm32-rs/stm32h7xx-hal #520: USB initialisation unreliable [fix included]

ryankurte, 2025-02-12:
Using an STM32H735 with USB the application runs correctly immediately after flashing or while a debugger is attached, however, without the debugger attached / on power cycling after a flash USB_OTG often fails to initialise, locking up the processor.

https://github.com/stm32-rs/synopsys-usb-otg/issues/37 proposes configuring interrupts prior to initialising the OTG peripheral, which appears to have resolved the issue for me.

```rust
// GAHBCFG.GINT = 1
cx.device.OTG_HS_GLOBAL.gahbcfg.modify(|_,w| w.gint().set_bit());
// GINTMSK.OTGINT = 1; GINTMSK.MMISM = 1;
cx.device.OTG_HS_GLOBAL.gintmsk.modify(|_,w| w.otgint().set_bit().mmism().set_bit());

let usb = USB1::new( ... )
```

Opening this issue to hopefully make it easier for others experiencing this to find the fix, and I think it would make sense to just fix this in `Usb::new_unchecked`?

richardeoin, 2025-02-16:
Hey, thanks for making an issue! I don't see an issue with including this in `Usb::new_unchecked`. In any case it won't generate a CPU interrupt unless the interrupt is also unmasked in the NVIC. Feel free to make a PR, or next time I'm working on this crate I'll pick it up.

ryankurte, 2025-07-08:
welp, turns out this only works below a certain clock frequency... there's a proposal for an upstream fix in https://github.com/stm32-rs/synopsys-usb-otg/issues/37 that seems to work for clock rates above 400MHz.